nordicsemi/nrfx #2: Doxygen warnings

JaskoWojtek, 2017-12-26:
Doxygen version 1.8.14

warning: Tag `SYMBOL_CACHE_SIZE' at line 350 of file `nrfx.doxyfile' has become obsolete.
warning: Tag `XML_SCHEMA' at line 1486 of file `nrfx.doxyfile' has become obsolete.
warning: Tag `XML_DTD' at line 1492 of file `nrfx.doxyfile' has become obsolete.

kl-cruz, 2018-02-07:
We would like to refactor the documentation a little in one of next releases. We will try to support new doxygen too.

smikai, 2020-01-08:
As there is an issue for doxygen warnings, I will update it.

Doxygen vesion 1.8.16 

warning: Tag 'PERL_PATH' at line 2186 of file 'nrfx.doxyfile' has become obsolete.
warning: Tag 'MSCGEN_PATH' at line 2208 of file 'nrfx.doxyfile' has become obsolete.

nika-nordic, 2021-11-24:
`nrfx` supports same Doxygen version as NCS project (1.9.2 currently) and for this particular version no warnings are produced in nrfx 2.6.0.

closed 2021-11-24 by nika-nordic