nordicsemi/nrfx #15: nrfx_gpiote: Possible RAM violation

ArnaudNe, 2018-04-04:
Hi,

Trying to use `nrfx_gpiote_in_uninit` on a non initialized GPIO causes RAM violation due to the returned value of `channel_port_get(pin)` (`0xFF`).

See line **638** of `nrfx_gpiote.c` :

```C
    channel_free((uint8_t)channel_port_get(pin));
```

and line **220**

```C
static void channel_free(uint8_t channel_id)
{
    m_cb.handlers[channel_id] = FORBIDDEN_HANDLER_ADDRESS;
    if (channel_id >= GPIOTE_CH_NUM)
    {
        m_cb.port_handlers_pins[channel_id - GPIOTE_CH_NUM] = (int8_t)PIN_NOT_USED;
    }
}
```
The follwing instruction, with `channel_id` = `0xFF` causes the RAM to be overriden outside of the `m_cb.handlers` array range.

```C
m_cb.handlers[channel_id] = FORBIDDEN_HANDLER_ADDRESS;
```

kl-cruz, 2018-04-04:
Hi,
Overriden RAM is serious problem. We need to know more.
What do You mean non initialized GPIO? Could You provide some code causes this behaviour?
I just wonder if asserts placed at the top of each public visible functions in gpiote driver can prevent this kind of issue.
Thanks for it!

ArnaudNe, 2018-04-05:
I tried to use `nrfx_gpiote_in_uninit()` before having called `nrfx_gpiote_in_init()`.

This causes the `channel_port_get()` function to return `0xFF` (line 638 of `nrfx_gpiote.c`).

In the `channel_free()` function (line 220 of `nrfx_gpiote.c`), we finally execute `m_cb.handlers[channel_id] = FORBIDDEN_HANDLER_ADDRESS;` with channel_id being the value returned by `channel_port_get()` (0xFF) and that is outside of the array bounds.

kl-cruz, 2018-04-09:
I checked it and problem can occur when assertions are disabled. Did You try to enable assertions?

We assumed that working with unintialized pin is caused by issue in architecture design:

When `nrfx_gpiote_init` is executed before working with other functions it sets all pins as uninitilized:
https://github.com/NordicSemiconductor/nrfx/blob/master/drivers/src/nrfx_gpiote.c#L236
All of the gpios are configure with value NOT_USED (-1):
https://github.com/NordicSemiconductor/nrfx/blob/master/drivers/src/nrfx_gpiote.c#L153
And then uninitialized GPIO should be detected
https://github.com/NordicSemiconductor/nrfx/blob/master/drivers/src/nrfx_gpiote.c#L618

We will consider to change it to runtime check.

nika-nordic, 2021-11-24:
It was changed to runtime check (despite user should not attempt to uninitialize uninitialized pin)

closed 2021-11-24 by nika-nordic