nordicsemi/nrfx #31: [not a bug] Example projects

dpiegdon, 2018-06-14:
Hi,
I was really missing some example project using this nrfx repository.
So I created one working with arm-none-eabi-gcc and make.
Just wanted to let you know: https://github.com/dpiegdon/nordic-template .
Seems to work fine (I added the gpiote example from the SDK), but I would be glad about any suggestions!

Cheers!

dazhbog, 2020-07-30:
@dpiegdon I just started using your template, thanks! Literally the only NRFx quick start example with CMSIS that i could find.

It would be lovely if someone from Nordic checked it and/or had a more up to date version. @kl-cruz @anangl

dpiegdon, 2020-07-31:
Thanks!
I'd love to upgrade to a more recent version of nrfx, but there were some conflicts between recent 802.15.4 drivers and nrfx versions. I was told they are currently in the process of weeding out and it might take a bit. See my pull requests at https://github.com/NordicSemiconductor/nRF-IEEE-802.15.4-radio-driver
None the less, I've been using the setup repeatedly and like it very much.
NB, check out the different branches or https://github.com/dpiegdon/dpsgo for more examples.

nika-nordic, 2022-11-22:
Hi @dpiegdon , @dazhbog ,

Just wanted to let you know that we introduced samples for `nrfx` drivers here: https://github.com/zephyrproject-rtos/hal_nordic/tree/master/nrfx/samples .
Documentation can be found here: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fstruct_drivers%2Fstruct%2Fnrfx_examples.html&cp=9_1

Samples use Zephyr build system for ease of use, but they can be very easily ported to bare-metal environment.