nordicsemi/nrfx #43: Make nrfx_<peripheral>_uninit() be idempotent

dhalbert, 2018-10-13:
Right now nearly all the nrfx device driver routines carefully check to make sure the driver instance is in the right state. `nrfx_*_init()` makes sure the instance is in state `NRFX_DRV_STATE_UNINITIALIZED`, and returns an error code if it's not in that state.  Almost other all routines make sure it is _not_ in state `NRFX_DRV_STATE_UNINITIALIZED`, by doing an `NRFX_ASSERT(state != NRFX_DRV_STATE_UNINITIALIZED)`, directly or indirectly. Most notably, even `nrfx_*_uninit()` does this check.

Unfortunately, this means that if you have a driver instance whose state is unknown, there's no simple way to force it it to be uninitialized. If you try to `uninit()` you risk causing a fatal error if `NRFX_ASSERT()` is enabled (e.g., in an `NDEBUG` build). A somewhat awkward workaround is to always `init()` the instance first, get an error return if it's already initialized, and then `uninit()`. But this might cause side effects from `init()`-ing that you don't want to happen.

This has come up for us (https://github.com/adafruit/circuitpython) because we want to be able to do a soft reset of our application and set all our driver instances to uninitialized, without having to track their state.

I propose that `uninit()` be made idempotent, so it can safely be called on an instance that's in any state. If you call `uninit()` on an instance that's already in state `NRFX_DRV_STATE_UNINITIALIZED`, it will simply return.

I have a PR already made up to do this. I haven't submitted it yet, but will do so if you like this idea.

An alternative is to add API calls that return the state, but that's more work. Making `uninit()` be idempotent is simpler, I think.

Thanks for thinking about this.

dhalbert, 2018-10-15:
Also being discussed in https://devzone.nordicsemi.com/f/nordic-q-a/39433/nrfx_-_uninit-not-idempotent-can-t-be-safely-called-on-an-already-uninitialized-driver-instance

kl-cruz, 2018-11-05:
Hi, thanks for this.
Could you explain how the soft reset mechanism works?
We designed our drivers to use init-uninit functions in pair. That's the reason why we are using assert to check if driver is in the right state.

dhalbert, 2018-11-05:
In CircuitPython, the user's Python program can assign pins dynamically to various peripherals. For instance, a pin may be allocated to I2C, SPI, GPIO, etc., or nothing. The user's program may throw a Python exception, or simply be interrupted via the keyboard etc. We want to be able to restart CircuitPython without a hard reset, without forcing USB re-enumeration, etc. But we do need to reset all the pins and drivers to a known quiescent state. If we can't call uninit() without knowing whether the driver is marked as initialized, then we can't do the soft reset.

We could have boolean inited/uninited flags for all the nrfx drivers we use, and track them that way, but the state of the driver is already known to the driver, though hidden inside the driver data structure. So an API addition that either allows uninit() to be called when the state is unknown, or an API that returns the state of the driver would be helpful.

kl-cruz, 2018-11-26:
API extension which allows to to return driver's state seems reasonable. I will open task for it to one of the future releases.

SebastianBoe, 2022-08-22:
Hi, I work with chain-loaded images where each image does not trust that the image before it has unitialized it's drivers before booting the next one and would also appreciate an API for uninitializing a driver without knowing the state of the driver.

dhalbert, 2024-01-17:
The 3.2.0 release has added functions like `nrfx_twim_init_check()`, which return true or false based on whether the driver was initialized or not. This satisfies the alternative solution proposed in https://github.com/NordicSemiconductor/nrfx/issues/43#issuecomment-441665189. Thank you.

@SebastianBoe looks like this will help you too.

closed 2024-01-17 by dhalbert

SebastianBoe, 2024-01-18:
It doesn't help for the bootloader usecase actually, as nrfx_twim_init_check will return that the driver is uninitialized even though the bootloader initializes the driver.

dhalbert, 2024-01-18:
@SebastianBoe I can reopen this. It looks like you work for Nordic, so do you have an internal issue open?

SebastianBoe, 2024-01-19:
I do not, but there is some context in https://nordicsemi.atlassian.net/browse/NRFX-2237

dhalbert, 2024-01-19:
I don't have access to that Jira. I'd assume it was Nordic internal only.