nordicsemi/nrfx #46: unable to get nrfx_nfct to detect field lost on 82540

PeterKenyon, 2018-10-24:
I have been struggling to get the nrfx_nfct driver to do anything sensible on my 82540 board, I don't suppose there is any example code or test code available, I get a field detected once and then nothing, im only trying to get field detected and field lost events

kapi-no, 2018-10-25:
Hi,

>  I don't suppose there is any example code or test code available

Unfortunately, we don't have any example code that shows how to use the NFCT driver yet. However, you can share what you have done so far, so it's easier for me to help you with your issue.

> I get a field detected once and then nothing, im only trying to get field detected and field lost events

Note that the NFCT driver uses additional HW resource: NRF_TIMER (4th instance by default) as a workaround for its HW anomalies. You should not use this timer for other purposes.

You need to make sure that HFCLK is running before you activate the NFCT peripheral with ```nrfx_nfct_state_force(NRFX_NFCT_STATE_ACTIVATED)```. However, this is irrelevant in your case if you only want to track field detected and field lost events and you don't need to exchange any data over NFC.

PeterKenyon, 2018-10-25:
Thanks for the prompt reply, I have enabled TIM4 (sdk_config), and I do see it fire (once), I am adding a lot more NRFX_DEBUG statements in nrfx_nfct.c, so work in progress, there are a lot of workarounds, (us software chaps have to cope with ;-( ), Ill keep you updated and maybe try to get a PCA10056 board SES project, the hardware Im using is a custom board

PeterKenyon, 2018-10-25:
Here is a small ses project that demonstrates the problem, nb there is a project macro that points to the SDK
[pca10056-nfc.zip](https://github.com/NordicSemiconductor/nrfx/files/2514323/pca10056-nfc.zip)

PeterKenyon, 2018-10-25:
Actually I have decided to adopt a different approach and just put the cpu into POWER OFF and use nfc to wake it up, this works well with the NRF52840, so I will put the rest of the nfc exploration into hold for now, cheers PK

PeterKenyon, 2018-10-25:
you are a genius sir, thanks a bunch

closed 2018-10-26 by PeterKenyon