nordicsemi/nrfx #51: NFCT example

hoihu, 2019-01-17:
We are using the nrfx library within micropython and have worked on integrating nfc functionality, without linking the t4t binary.

Unfortunately we are stuck on how to use the driver to  emulate eg a simple url tag. We have raised this already here https://devzone.nordicsemi.com/f/nordic-q-a/41705/trouble-with-nrfx-nfct-driver

but the response was not convincing.

So my question is if somebody could point us to sample code that uses the driver. 

So far we were able to detect FIELD, switch on HCLK and the received a read error event.

My feeling is that we are really close to getting the driver to work so it would be a pitty if we have to abandon this development path.

Thanks for any feedback

Martin

kapi-no, 2019-01-17:
Hi,

The nrfx_nfct driver cannot emulate a URL tag alone. It's a low-level module for NFC stack (a counterpart of legacy hal_nfc_t2t module from the nRF5 SDK) and requires a module on top of it, which can interpret NFC frames at the Type 2 Tag level and respond to them (e.g. nfc_t2t_lib from the nRF5 SDK). However, the NFC libraries from nRF5 SDK are not compatible with the nrfx_nfct driver yet.

kapi-no, 2019-01-23:
You need to be able to interpret incoming commands at the Type 2 Tag protocol level and generate appropriate responses.

For your use case, you need to allocate a memory block with tag's memory content (T2T header content + NDEF part) and respond to the subsequent T2T read commands which will request a specific area of this memory until the whole content is read by the NFC poller.

The driver was tested at the NFC-A protocol layer to check if it is capable of detecting field and exchanging data at the NFC-A level. As a rule of thumb, any RX callbacks from the nrfx_nfct driver should be followed by a response generated by the driver's user.

qdung, 2019-02-22:
> We are using the nrfx library within micropython and have worked on integrating nfc functionality, without linking the t4t binary.
>
> Unfortunately we are stuck on how to use the driver to emulate eg [...]

Hi Martin, can you share me how to use NFCT driver to read NDEF from Android phone? I'm build a library of nRF52 NFC for Arduino IDE

hoihu, 2019-02-23:
No I can‘t since we didn‘t want to spend the additional development time for a pure nrfx nfc driver (which was the prefered way of integrating it into uPy)

so your best option is likely to use the t2t library with the nordic SDK (where there are lots of precompiled examples available too)

hoihu, 2019-09-14:
@kapi-no Are there any updates on this? Are there any plans from nordic's side that the NFCT driver is supported by an open source Tag2/4 module?

kapi-no, 2019-09-16:
@hoihu
No, there are no plans to provide open-source Tag libraries in Nordic currently.

However, the Tag libraries in the nRF SDK are compatible with the NFCT driver from nrfx now (since v15.3.0 nRF5 SDK). We also plan to support this compatibility in future releases.

closed 2019-09-23 by hoihu