nordicsemi/nrfx #52: Some I2S master clock frequencies were removed from nrf52840_bitfields.h in release 1.6.0

jtomaszb, 2019-01-31:
Master clock frequency defines for divider values 6 through 2 were removed from the nrf52840_bitfields.h header file in release 1.6.0. This is an issue for us because we have been using I2S_CONFIG_MCKFREQ_MCKFREQ_32MDIV4 in our configuration.

We can add these defines back in our codebase, as we know it is supported by the NRF52840, but I would like to understand why they were removed. Is there an underlying issue that was the root cause for this change?

dhalbert, 2019-02-04:
Just to be clear, is the new official maximum is 4MHz (4000 kHz)? I wasn't sure what was going to be removed from the documentation. Thanks.

jarz-nordic, 2019-02-05:
@dhalbert : Yes.

kl-cruz, 2019-05-17:
Documentation was updated: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf52840%2Fi2s.html&cp=3_0_0_5_10_9&anchor=register.CONFIG.MCKFREQ
I'm closing this issue.

closed 2019-05-17 by kl-cruz