nordicsemi/nrfx #53: High power consumption when using nrfx_qspi in blocking mode

SuhaibI, 2019-03-01:
Hello, I've found that the QSPI driver causes high power consumption when used in blocking mode (initialized with NULL as a handler.) It seems this is because `nrfx_qspi_cinstr_xfer` calls the `nrf_qspi_int_enable` function after the transfer is finished. Any generated interrupts never trigger a handler, since `NRFX_IRQ_ENABLE` isn't called in the init function in the non-blocking case. Uniniting the qspi driver doesn't help in this case, since the interrupt isn't cleared in the disable function. It seems the device doesn't go into sleep mode because of this unhandled interrupt, resulting in a current draw of around 3 mA.

I found 2 fixes for this issue. First, I removed the call to `nrf_qspi_int_enable` function in `nrfx_qspi_cinstr_xfer.` I don't think this is needed since `nrf_qspi_int_enable` is called when needed in `qspi_task_perform` anyway. My second fix was to add `NVIC_ClearPendingIRQ` in `NRFX_IRQ_DISABLE` (in nrfx_glue.h) after the call to `NVIC_DisableIRQ.` This recovers the power consumption state upon uniniting, even without the first fix. I'm wondering if this is good practice to have in general, or whether it might cause any other issues.

Can you comment on the issue and fixes, and whether they might have some negative repercussions? Thanks!

kl-cruz, 2019-03-04:
Hi SuhaibI!
I'd like to start with second fix. I do not recommend to change interrupt-related functions in `nrfx_glue.h` file. It can cause unpredicted behaviour of some drivers. 
I'll try to test your first solution and prove that it can help with the issue. The answer will be ready in a few days! 
I'd be grateful if you could provide me with the example code of this issue.
Thanks!

SuhaibI, 2019-03-08:
Hi Karol,

Yeah that's fair, I suspected that modifications to `nrfx_glue.h` might cause issues in other drivers. 

It's hard for me to provide example code as our code is tied up with a lot of proprietary firmware and is specific to the external flash chip we're using on our custom board. The following steps should reproduce the issue:

- Perform a QSPI initialization in blocking mode (pass NULL as the handler into `nrf_drv_qspi_init`)
- Perform chip-specific initialization (send custom instructions as required)
- Perform a single device read/write/erase operation

It is necessary to try to enter sleep mode by running `nrf_pwr_mgmt_run`. This is not currently the case in the QSPI example for the PCA10056. Let me know if need more information to reproduce the issue.

Thanks!

kl-cruz, 2019-07-26:
Fixed in nrfx 1.7.2:
https://github.com/NordicSemiconductor/nrfx/commit/13a7de7de72cfd444af468f11c0dbb9db7a26172#diff-f3f74861f7809f4c37c375c2c346cc11L196
 Thank you for for finding it!

closed 2019-07-26 by kl-cruz