nordicsemi/nrfx #54: nrfx_qspi silently truncates read/writes that aren't multiples of 4

Xenoamor, 2019-03-20:
If nrfx_qspi_read or nrfx_qspi_write is called with a size that isn't a multiple of 4 it is silently truncated. This appears to be a hardware limitation although I feel the sourcecode should make this known

kl-cruz, 2019-05-16:
That's right:
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf52840%2Fqspi.html&cp=3_0_0_5_18_9_12&anchor=register.READ.CNT
I'll create internal ticket for this. Thank you!

kl-cruz, 2019-07-26:
Hi,
nrfx 1.7.2 was released with changes related to this:

https://github.com/NordicSemiconductor/nrfx/commit/13a7de7de72cfd444af468f11c0dbb9db7a26172#diff-f3f74861f7809f4c37c375c2c346cc11R353

https://github.com/NordicSemiconductor/nrfx/commit/13a7de7de72cfd444af468f11c0dbb9db7a26172#diff-f3f74861f7809f4c37c375c2c346cc11R369

Thank you for finding it!

closed 2019-07-26 by kl-cruz