nordicsemi/nrfx #68: Question/documentation: what does MDK stand for?

mokus0, 2020-02-04:
It's used all over and I know what it _does_ but I can't find an explanation of why it's called "mdk" (which might be helpful in documenting my own build rules, to explain at point of use why it's being pulled in)

adfernandes, 2020-02-06:
Really? I always thought it was `Middleware Development Kit` since "middleware" is usually the Hardware Abstraction Layer (HAL) plus a bit of driver abstraction over the "raw register" access level...

I think that [Microcontroller Development Kit](http://www2.keil.com/mdk5/) refers to Keil's IDE product, not the "middleware" of Nordic's software in this repo.

kl-cruz, 2020-02-07:
Hi,
@inductivekickback You are right, It is Microcontroller Development Kit. You can read more at: https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-MDK

adfernandes, 2020-02-07:
Darn. Another good theory ruined by facts! 😉

closed 2020-08-21 by kl-cruz