nordicsemi/nrfx #69: nRF5340 Audio clock support

koffes, 2020-02-05:
There are some desired features in the ACLK driver/nrf_clock.h. (Tested by using the I2S module).

nrf_clock_hfclkaudio_config_set should have upper and lower bounds on the allowed write values as per nRF5340_OPS_v0.5.1.pdf. In addition, there should be a limiter to ensure that the user does not request too large step changes from the PLL.

In order to use the PLL, One needs to write to the NRF_CLOCK->TASKS_HFCLKAUDIOSTART register.
There should be an API for this.

kl-cruz, 2020-02-07:
Hi @koffes !
I checked this functionality. Second argument of the `nrf_clock_hfclkaudio_config_set` function takes parameter stored at two bytes:
https://github.com/NordicSemiconductor/nrfx/blob/master/hal/nrf_clock.h#L536
It should cover all values supported by this register:
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fps_nrf5340%2Fchapters%2Fclock%2Fdoc%2Fclock.html&cp=3_0_0_3_8_2_44&anchor=register.HFCLKAUDIO.FREQUENCY
I see in the documentation text that there are two frequency bands. I assume that these are the boundaries that you are talking about. I will ask developers to explain if other frequencies, outside these bands are supported.

To start audio PLL you can use nrfx_clock API. There is dedicated function that starts clock at desired domains: `nrfx_clock_start(NRF_CLOCK_DOMAIN_HFCLKAUDIO)`. 
If you use hal functions directly you can call:
```
nrf_clock_event_clear(NRF_CLOCK, NRF_CLOCK_EVENT_HFCLKAUDIOSTARTED);
nrf_clock_int_enable(NRF_CLOCK, NRF_CLOCK_INT_HFAUDIO_STARTED_MASK);
nrf_clock_task_trigger(NRF_CLOCK, NRF_CLOCK_TASK_HFCLKAUDIOSTART);
```
Remember to check `NRF_CLOCK_EVENT_HFCLKAUDIOSTARTED` event in while loop or interrupt whether audio clock is started or not.

koffes, 2020-03-29:
Thanks for the reply. I see now that this API was added to nrfx just a few days before this ticket was created. I believe nrfx 2.1.0 was not in the Zephyr manifest at that time. That's why I did not see it. 
The missing pieces then seems to be the bounds on the upper an lower values and step size. If responsibility of using legal values should rest with the user, then this ticket can be closed.

closed 2021-03-18 by nika-nordic