nordicsemi/nrfx #80: NRFx mismatch with SDK NRFx changelog or version?

dazhbog, 2020-07-30:
Hi,

Inside NRF SDK 17, in the _modules/nrfx_ directory there is this change log:

```
## [1.8.4] - 2020-04-27

## [1.8.3] - 2020-03-20

## [1.8.2] - 2020-03-05

```

But in this repo there are different entries with dates also being in 2020. Am i looking at the same project??

```
## [2.2.0] - 2020-04-28

## [2.1.0] - 2020-01-24

## [2.0.0] - 2019-11-06

## [1.8.1] - 2019-10-21
```

Which NRFx source shall we use? 

How is the NRFx development going vs. NRF connect and your nrf_drv legacy drivers? It is really confusing what to choose.

Thanks!

dazhbog, 2021-01-22:
@nika-nordic Hi, can you have a look at this or do you need more information from my side?

Thanks!

nika-nordic, 2021-01-22:
@dazhbog Hi,

for nRF5 SDK we maintain a legacy branch of nrfx, versioned 1.x. This is API-compatible with nrf_drv_* API present in nRF5 SDK. New releases of this branch consist mostly of bug fixes and new MDKs.

nrfx 2.x is an actively developed branch where all new features can be found. It also supports new devices like nRF91 or nRF53. This one is integrated into Zephyr RTOS and NCS.

Please keep in mind that some APIs changed between nrfx-1.x & nrfx-2.x.

Regards,
Nikodem

closed 2021-03-18 by nika-nordic