nordicsemi/nrfx #82: nRF5340 TWIM @1mbps: driver sets incorrect gpio drive strength

haakonsh, 2020-11-02:
In nrfx_twim.c at line 75: 

#define TWIM_PIN_INIT(_pin) nrf_gpio_cfg((_pin),                     \
                                         NRF_GPIO_PIN_DIR_INPUT,     \
                                         NRF_GPIO_PIN_INPUT_CONNECT, \
                                         NRF_GPIO_PIN_PULLUP,        \
                                         NRF_GPIO_PIN_S0D1,          \
                                         NRF_GPIO_PIN_NOSENSE)

We need a drive strength of NRF_GPIO_PIN_E0E1 according to nRF5340 Prod Spec's [Pin assignments](https://infocenter.nordicsemi.com/topic/ps_nrf5340/chapters/pin.html?cp=3_0_0_8_0):

"TWI - For the fastest TWI 1 Mbps mode, the two high-speed TWI pins must be configured in the TWI peripheral's PSEL registers, and the 20 mA open drain driver enabled using the E0E1 drive setting in the DRIVE field of the PIN_CNF GPIO register."

nika-nordic, 2020-11-04:
Hi,

you could you try to call:
```
nrf_gpio_cfg(<pin>,
             NRF_GPIO_PIN_DIR_INPUT, 
             NRF_GPIO_PIN_INPUT_CONNECT,
             NRF_GPIO_PIN_PULLUP, 
             NRF_GPIO_PIN_E0E1,
             NRF_GPIO_PIN_NOSENSE)
```

after `nrfx_twim_init` as workaround.

Please remember that according to `Pin assignments` doc section only P1.2 and P1.3 are compatible with TWIM 1 Mbps mode.

nika-nordic, 2021-05-05:
Added support for TWIM 1 Mbps mode on nRF5340 in latest release (2.5.0)

closed 2021-05-05 by nika-nordic