nordicsemi/nrfx #83: nrf_radio_power_set() use directly a bool to write a register

nbrunner, 2020-11-20:
in [(https://github.com/NordicSemiconductor/nrfx/blob/master/hal/nrf_radio.h#L1896] use a bool to directly write the register but bool is not guaranteed to have a 0 or 1 value. It should be like the function nrf_radio_modecnf0_set() that also use a bool.

p_reg->POWER = radio_power ? (RADIO_POWER_POWER_Enabled << RADIO_POWER_POWER_Pos) : (RADIO_POWER_POWER_Disabled << RADIO_POWER_POWER_Pos);

nika-nordic, 2020-11-23:
Hi,

this is how `nrf_radio_power_set` function declaration looks like:
```
NRF_STATIC_INLINE void nrf_radio_power_set(NRF_RADIO_Type * p_reg, bool radio_power);
```
User is expected to provide value `true` or `false` as argument for the `radio_power` parameter. `nrfx` follows C99 standard and in section 7.16 of this document you can find following statement:
```
(...) They are `true` which expands to the integer constant 1, `false` which expands to the integer constant 0 (...)
```
This regards symbols from `<stdbool.h>` header that is used by `nrfx`.

As long as user uses the API as intended I see no risk. Nevertheless, any non-zero value can be interpreted as `true`, so we will consider to rework the assignment as suggested.

Thank you for your input!

nbrunner, 2020-11-23:
Thank you for considering rework.
Currently zephyr don't directly use true or false because of that:
[](https://github.com/zephyrproject-rtos/zephyr/issues/30130)

nika-nordic, 2021-03-18:
Rework will be included in upcoming nrfx release. Thank you for pointing this out

closed 2021-03-18 by nika-nordic