nordicsemi/nrfx #88: qdec without reportrdy interrupt - nRF52840

jpfaff, 2021-05-20:
I was trying to use the qdec driver (through the zephyr driver) without using interrupts by setting NRF_QDEC_EVENT_REPORTRDY_DISABLED. The idea is to retrieve the ACC value only when manually "fetching" and "getting".
Doing this I always got at most 1(-1) when reading the ACCREAD register.
It seems to be because of this line: https://github.com/NordicSemiconductor/nrfx/blob/b5399066bd7f3dc32ee15510d06ef7137bcacf36/drivers/src/nrfx_qdec.c#L121
The short SHORT_REPORTRDY_READCLRACC clears ACC and sets ACCREAD automatically at each REPORTRDY event, which seems to happen even with NRF_QDEC_EVENT_REPORTRDY_DISABLED, but no interrupt is called. Therefore the counted steps are lost.
Is there anything I missed?

nika-nordic, 2021-06-11:
Hi @jpfaff,

looks like I cannot answer this immediately, but will try to look into the driver implementation and come back to you

nika-nordic, 2021-08-09:
Hi again @jpfaff,

I've finally found some time to look into this and there are my findings. Looks like current implementation assumes that  `NRF_QDEC_EVENT_REPORTRDY_DISABLED` prevents the REPORTRDY event to be generated which is a wrong assumption. I think we can move the enabling of shorts underneath this `if`: https://github.com/NordicSemiconductor/nrfx/blob/master/drivers/src/nrfx_qdec.c#L134 . Then accumulated samples could be collected via `nrfx_qdec_accumulators_read`.

For now the workaround would be to simply call `nrf_qdec_shorts_disable(NRF_QDEC, NRF_QDEC_SHORT_REPORTRDY_READCLRACC_MASK);` after QDEC Zephyr driver initialization. Let me know whether this helps

jpfaff, 2021-09-01:
Hi @nika-nordic,
Sorry for the late reply. 
I ended up copying the driver (creating a custom one) and commenting out the shorts_enable line.
Moving the shorts enabling inside the if, as you describe, should indeed solve the problem. I just can't test this right now.
Thank you.

nika-nordic, 2021-11-24:
Fixed in nrfx 2.6.0

closed 2021-11-24 by nika-nordic