nordicsemi/nrfx #90: nfrx_power and nrfx_clock coexistence issue

Yagoor, 2021-06-26:
I found a possible bug when initialising the clock and the power drivers.
If you initialise the clock driver without a handler and initialise the power driver.
Since they share the same interrupt, at least on the nrf52840, the interrupt handler will call the power driver handler and then call the clock driver handler without checking if the handler is there. 
The power driver always has the interrupt handler but on the clock you don't need to have the handler.

nika-nordic, 2021-07-12:
Hi @Yagoor
thank you for your report. We will check the issue internally

nika-nordic, 2021-08-09:
Hi again @Yagoor,

according to the current implementation this statement is wrong: `he power driver always has the interrupt handler but on the clock you don't need to have the handler.`

Event handler in `nrfx_clock_init` is checked with assertion: https://github.com/NordicSemiconductor/nrfx/blob/master/drivers/src/nrfx_clock.c#L181
so with assertions enabled passing NULL handler will cause an error.

Therefore the user is expected to provide a valid event handler during initialization.

Yagoor, 2021-08-09:
Hi @nika-nordic,

Thank you for your reply.

I missed the assert there. 
On our Contiki-NG port we defined out the asserts that is why I didn't see this. 
Thank you for your input.

I'll close the issue. :)

closed 2021-08-09 by Yagoor