nordicsemi/nrfx #91: TWIS broken in nrfx 2.5 ( ncs v1.6 )

zoon81, 2021-07-08:
Dear Nordic,

Seams like the twis driver in nrfx2.5 is broken. I just updated from ncs v1.5 and my twis driver not working properly.
The issue is that the driver only sending every second bytes from my buffer. I went back to ncs v1.5 and there is no such a problem over there.
Am I missed something about the migration from ncs 1.5 to ncs 1.6?
I am looking forward to hear about this issue.
Thank you.

I am using the nRF9160_DK for testing with Zephyr OS build v2.6.0-rc1-ncs1

nika-nordic, 2021-07-12:
Hi @zoon81,
are you using nrfx_twis API directly or some other Zephyr API? This info would help us narrow down the source of the issue

nika-nordic, 2021-08-09:
Hi again @zoon81,

I've double-checked our nrfx tests for TWIS on nRF9160 DK and they are all passing. Are you using nrfx_twis API directly or some Zephyr API? If you use Zephyr API then the issue is likely outside of the nrfx and should be reported on `Nordic DevZone`

closed 2021-11-24 by nika-nordic