nordicsemi/nrfx #92: `nrf9160.svd` is malformed

jonas-schievink, 2021-08-06:
The SVD file for the nRF9160 (https://github.com/NordicSemiconductor/nrfx/blob/master/mdk/nrf9160.svd) contains 3 errors:

In lines 628, 635, and 4841, a register access value of `read-writeonce` is specified, but it should be `read-writeOnce`, with an uppercase O.

The valid access types are specified at the bottom of this page: https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_special.html#elem_access

nika-nordic, 2021-08-09:
Hi @jonas-schievink ,

thank you for raising this issue. We will look into this

nika-nordic, 2021-08-09:
Hi again @jonas-schievink 

looks like there is a bug in our .svd file creation process. The fix should be available in next release. Thank you again for reporting this

jonas-schievink, 2021-08-09:
Thanks for the quick response!

nika-nordic, 2021-11-24:
New nrfx release 2.6.0 contains updated MDK. Closing

closed 2021-11-24 by nika-nordic