nordicsemi/nrfx #113: intentional warning about uint16_t?

MegaC-C, 2023-06-13:
https://github.com/NordicSemiconductor/nrfx/blob/1c721175f22dbb1bf125a570a427b53f810881bb/hal/nrf_pwm.h#L95

^this gives me a compile time warning complaining about incompatibly of uint16_t and my used array of type nrf_pwm_values_grouped_t (I m using nRF Connect SDK 2.2.0)

however when parentheses are used for the sizeof
>   #define NRF_PWM_VALUES_LENGTH(array)  (sizeof(array) / (sizeof(uint16_t)))

 the warning is surprised, is there are reason why is this not used in the first placed?

MegaC-C, 2023-06-15:
Thank you, @nika-nordic, for your prompt response! I apologize for overlooking the link to https://github.com/NordicSemiconductor/nrfx/issues/105; it addresses the exact problem I encountered.

However, I believe it is sufficient to enclose sizeof(uint16_t) in parentheses as shown below:
`#define NRF_PWM_VALUES_LENGTH(array) (sizeof(array) / (sizeof(uint16_t)))`

This adjustment ensures that the division operation is executed correctly. The addition of parentheses corrects the order of precedence, as the `sizeof` operator without parentheses  takes precedence over the division operator. Am I correct in my understanding?

awojasinski-nordicsemi, 2023-06-28:
Hi @MegaC-C,

the new nrfx 3.1.0 release fixes a reported issue.
I want to thank you on behalf of the team working on nrfx.

closed 2023-06-28 by awojasinski-nordicsemi