nordicsemi/nrfx #121: `nrfx_timer_extended_compare` fails assertion when timer is paused

Yaxit, 2024-06-07:
I'm working on an application with timers and PPI.
Today I discovered that when using the function `nrfx_timer_extended_compare` with a timer enabled but **paused** I get a runtime assertion error.
This assertion fails, as the timer state is `NRFX_DRV_STATE_POWERED_ON` and not `NRFX_DRV_STATE_INITIALIZED` https://github.com/NordicSemiconductor/nrfx/blob/4620e7ccfebc08dbd4c039bc7d530b62c4c68ca8/drivers/src/nrfx_timer.c#L320

A few lines above the function `nrfx_timer_compare` uses a different assertion, just validating that the state is not `NRFX_DRV_STATE_UNINITIALIZED`
https://github.com/NordicSemiconductor/nrfx/blob/4620e7ccfebc08dbd4c039bc7d530b62c4c68ca8/drivers/src/nrfx_timer.c#L293

Is this a desired behavior or ar bug? Seems strange to me that one cannot update the compare value of the timer when the timer is paused.

Yaxit, 2024-06-07:
As a small update, I tried to skip the assertion and the timer seems to work as intended, I just need confirmation that this is indeed a bug.

nika-nordic, 2024-07-26:
hello, this is indeed a bug and will be fixed in upcoming release

nika-nordic, 2024-07-29:
nrfx release 3.6.0 was published. I believe it should have the fix applied. Can you confirm on your end, please?

Yaxit, 2024-07-29:
Did not test it, but the code looks correct for `nrfx_timer_extended_compare`, we can close.

https://github.com/NordicSemiconductor/nrfx/blob/075e9e711383bb3babfae4249fd59c8c8d882f72/drivers/src/nrfx_timer.c#L320

closed 2024-07-29 by Yaxit