nordicsemi/nrfx #123: mdk: missing CRYPTOCELL HAL as documented in newest product specifications

mguetschow, 2024-07-15:
The newest versions of the product specifications for [nRF52840](https://docs.nordicsemi.com/bundle/ps_nrf52840/page/cryptocell.html), [nRF5340](https://docs.nordicsemi.com/bundle/ps_nrf5340/page/cryptocell.html) and [nRF9160](https://docs.nordicsemi.com/bundle/ps_nrf9160/page/cryptocell.html) include the following changelog:

> CRYPTOCELL – Updated description

In practice, the documentation has been largely extended to include detailed information about the DMA registers to directly interact with the CRYPTOCELL peripheral, including example "cryptographic flow"s. Those use register names and values such as `NRF_CC_MISC` that are defined in the product specification, but still missing from the MDK as published on the [website](https://www.nordicsemi.com/Products/Development-tools/nRF-MDK/Download) and here within `mdk/`.

In particular, the SVD files and the automatically derived header files would need to be updated to include the information available in the product specification.

Is this on the current roadmap? If so, when can we expect the update to be released?

nika-nordic, 2024-07-26:
Hello,

MDK is delivered separately from nrfx drivers. I will ask internally about product spec updates being reflected in MDK definitions

nika-nordic, 2024-08-01:
Hello,

I have asked internally and there is some ongoing work to update MDK with relevant definitions. I will provide more details once I have them

nika-nordic, 2024-10-17:
Hello @mguetschow ,

nRF52840 MDK 8.67.0 available in nrfx 3.7.0 seems to have missing CRYPTOCELL definitions now. nRF5340 and nRF9160 are to be added in near future

nika-nordic, 2026-08-27:
Hello @mguetschow  ,

I see that MDK contains the following now:
```
$grep -rl NRF_CC_MISC
nrf91/nrf9160/nrf9160.h
nrf91/nrf9120/nrf9120.h
nrf53/nrf5340/nrf5340_application.h
nrf52/nrf52840/nrf52840.h
```

Does it mean this issue can be closed?