nordicsemi/nrfx #125: Build warning when building with clang 18

aescolar, 2024-09-17:
Building drivers/src/nrfx_dppi.c with clang for the nrf5340 results in the following warning:
```
drivers/src/nrfx_dppi.c
drivers/src/nrfx_dppi.c:65:47: error: shift count >= width of type [-Werror,-Wshift-count-overflow]
   65 | static nrfx_atomic_t   m_allocated_channels = DPPI_AVAILABLE_CHANNELS_MASK;
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/src/nrfx_dppi.c:56:17: note: expanded from macro 'DPPI_AVAILABLE_CHANNELS_MASK'
   56 |     ((uint32_t)(NRFX_BIT_MASK(NRF_DPPI_CH_NUM_MAX) & (~NRFX_DPPI_CHANNELS_USED)))
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/nrfx_common.h:139:60: note: expanded from macro 'NRFX_BIT_MASK'
  139 | #define NRFX_BIT_MASK(x) (((x) == 32) ? UINT32_MAX : ((1UL << (x)) - 1))
      |                                                            ^  ~~~
```

Note, this is the expansion of this macro as it will be compiled:
```
static atomic_t m_allocated_channels = ((uint32_t)(((((((32)) > ((((0)) > (0) ? ((0)) : (0))) ? ((32)) : ((((0)) > (0) ? ((0)) : (0))))) == 32) ? (4294967295U) : ((1UL << ((((32)) > ((((0)) > (0) ? ((0)) : (0))) ? ((32)) : ((((0)) > (0) ? ((0)) : (0)))))) - 1)) & (~(0 | 0 | 0))));
```
(Note the preprocessor does not handle ternary conditional expressions, it just expands them as is).

### Environment

clang version 18.1.3
Ubuntu 24.04
Versions:
* Zephyr main 23eca6362dce63e52fcf6825f772e7ee7ae5226e
* hal_nordic af7b21c4f875bea26689307daa8e52e9a8e8323c

### Steps to reproduce

In a vanilla Zephyr workspace,
1. `mkdir build && cd build`
2. `ZEPHYR_TOOLCHAIN_VARIANT=llvm  cmake -GNinja -DBOARD=nrf5340bsim/nrf5340/cpuapp ../samples/hello_world/ -DCONFIG_NRFX_DPPI=y `
3. ninja
4. See warning

aescolar, 2024-09-17:
A quick fix would be
```
diff --git a/nrfx/drivers/nrfx_common.h b/nrfx/drivers/nrfx_common.h
index 8d7a2a7..af27efd 100644
--- a/nrfx/drivers/nrfx_common.h
+++ b/nrfx/drivers/nrfx_common.h
@@ -136,7 +136,7 @@ extern "C" {
  *
  * @return Bit mask.
  */
-#define NRFX_BIT_MASK(x) (((x) == 32) ? UINT32_MAX : ((1UL << (x)) - 1))
+#define NRFX_BIT_MASK(x) (((x) == 32) ? UINT32_MAX : ((1ULL << (x)) - 1))
 
 /**
  * @brief Macro for returning size in bits for given size in bytes.
```

Katty53251, 2025-02-25:
55.596646